robbinjanssen/home-assistant-omnik-inverter #291: Support for Omniksol-3K-TL

TheoKusters, 2024-02-27:
## Detailed description
Currently I use the integration https://github.com/hultenvp/home_assistant_omnik_solar but that one isn't supported anymore. Could you add support for Omniksol-3K-TL?

## Context
I have to migrate away from https://github.com/hultenvp/home_assistant_omnik_solar

## Diagnostics
I tried connecting to the inverter over http using the js, json and html method. All return:

```
<HTML>
<HEAD>
    <TITLE>401 Unauthorized</TITLE>
</HEAD>
<BODY BGCOLOR="#cc9999">
    <H4>401 Unauthorized</H4>
    Authorization required.
</BODY>
</HTML>
```

In the current integration I need to specify the ip address, serial number and the port number (8899)

robbinjanssen, 2024-04-02:
Did you try connecting via the TCP option?

TheoKusters, 2024-04-14:
Thanks Robbin, I missed that option. Just tried it and it works fine!