robbinjanssen/home-assistant-omnik-inverter #303: Unsupported Omnik converter

rdeherder, 2024-05-07:
## Detailed description
I'm trying to add my Omnik converter. Two issues:
- I cannot connect using the HA setup. I get a "Cannot connect" error message. Perhaps this is because it needs authentication;
- When I log into the converter and try to fetch the js, json or status.html files I get a 404's back. Both on the internal and external ip address.

Converter info:
Type: Omniksol-2K-TL
Serial: 60222xxxx

Is this a known issue?

robbinjanssen, 2024-05-07:
Hi,

There have been others who have successfully added the same inverter, see https://github.com/robbinjanssen/home-assistant-omnik-inverter/issues/15

Try looking in your network tab using the developer tools from where your inverter webpage fetches the data

rdeherder, 2024-05-09:
Hi.

I looked at #15 and I saw a newer settings page than my inverter has:

[image]

That probably means he/she has a newer inverter and would explain why trying to load the /js/status.js and /status.json files all end up in a 404 here.

The network tab when logging into my inverter:
[image]

On the plus side, I have been able to get data from the inverter using the [Inverter-Data-Logger](https://github.com/XtheOne/Inverter-Data-Logger) package. Data is retreived from the inverter and sent to my account on pvoutput.org. HA then reads the data from there. Works fine, but since this is cloud based I prefer to use your solution.

robbinjanssen, 2024-05-13:
You need to find out where your stats page on your inverter fetches the data from, might be a call to a JSON file in the network tab, or a JS file on page load.

If we know what the file and content is, we can add the support :)

MarijnS95, 2024-05-13:
@rdeherder I'm quite sure I/we have have seen this settings page before. Have you tried the TCP backend option (which doesn't seem to be mentioned, only the file-base option)?

MarijnS95, 2024-05-13:
Might have been https://github.com/robbinjanssen/home-assistant-omnik-inverter/issues/117, which also mentions TCP but before it was implemented in this package.

rdeherder, 2024-06-13:
> @rdeherder I'm quite sure I/we have have seen this settings page before. Have you tried the TCP backend option (which doesn't seem to be mentioned, only the file-base option)?

I don't know what you mean with TCP backend option. Can you elaborate?

MarijnS95, 2024-06-13:
@rdeherder this should be an option in the config flow when selecting the "Data source" from the dropdown.

rdeherder, 2024-06-14:
Ok, that worked! Thanks very much for your help.

closed 2024-06-14 by rdeherder