robbinjanssen/home-assistant-omnik-inverter #317: Integration not loaded in version 2024.7.1

Floriszz, 2024-07-08:
## Detailed description
The error says that something wrong about import_module.
Same as other integrations HACS and native;
Google Nest: https://github.com/home-assistant/core/issues/121521
HACS pod_point : https://github.com/home-assistant/core/issues/121284

## Context
The integration won't load anymore after update from 2026.6.4 to 2024.7.1

## Diagnostics
Sorry, because I did a restore I can not get to the logs anymore

klaasnicolaas, 2024-07-08:
Think I know what this is about, I'll make a bug fix

./Klaas

klaasnicolaas, 2024-07-08:
Without logs it becomes difficult to determine where the problem is and how to solve it.

./Klaas

Floriszz, 2024-07-10:
Sorry. Will update on my secondairy new rpi when it arrives🤞
It seems related to this issue;
https://github.com/home-assistant/core/issues/119223

robbinjanssen, 2024-07-12:
Seems like a fix is incoming: https://github.com/home-assistant/core/pull/121782

robbinjanssen, 2024-07-24:
Assuming this is fixed, i'm closing this issue

closed 2024-07-24 by robbinjanssen