robbinjanssen/home-assistant-omnik-inverter #327: Just a question about a specific piece of code

davidebeatrici, 2024-09-10:
I'm using this project as a reference for my own custom component, which integrates with a BMS.

There is a specific piece of code that bothers me: https://github.com/robbinjanssen/home-assistant-omnik-inverter/blob/f35a7ad1c05ea8eebd7c27dff6cf9e01b4d9b839/custom_components/omnik_inverter/coordinator.py#L35-L74

A `ConfigEntry` variable called `config_entry` is declared as an uninitialized class member.

The constructor (`__init__`) gets a `ConfigEntry` variable as an argument. It then proceeds to use that for the super (upper class) constructor.

But from that point onward the function accesses the member variable `config_entry` without ever assigning/initializing it.

I also thought `config_entry` could be part of the super class and the redeclaration is superfluous, but that wouldn't make sense because:

1. It's already passed as an argument to the constructor.
2. It would get overridden by the variable (of the same name) in the derived class.

Please note that I'm not a Python expert by any means and thus there are definitely concepts I'm not aware of.

However, I decided to open an issue here to ask the developer(s) directly and just in case this is effectively a bug.

davidebeatrici, 2024-10-11:
@robbinjanssen

davidebeatrici, 2024-11-12:
@robbinjanssen

robbinjanssen, 2024-11-13:
I think you're right, there's a specific piece of code in the home assistant core that sets the [`self.config_entry`](https://github.com/home-assistant/core/blob/dev/homeassistant/helpers/update_coordinator.py#L90-L93) for all classes extending the `DataUpdateCoordinator`.

So it _does_ get initialized but not at the place where you would expect it. It would be better to rename `entry` to `config_entry` I guess and pass it through to the `super().init`

I haven't touched this component in ages and don't really use it anymore, but PRs are welcome :-)

davidebeatrici, 2024-11-13:
Thank you for explaining, it's all clear now.

I don't have the hardware to test the integration, but I can create a pull request anyway. :)

closed 2024-12-22 by github-actions[bot]