robbinjanssen/home-assistant-omnik-inverter #347: MI600V2/MY-600

HA-Ath0n, 2025-01-04:
## Detailed description

## Context

## Diagnostics

```json
"Copy here the code or upload the file"
```

robbinjanssen, 2025-01-06:
Please open up a relevant issue containing a question or a problem.

MarijnS95, 2025-01-06:
I can only assume that this question is about supporting the v2 version of the BW-MI600 (the latter is in our support matrix). Since the user didn't ask this explicitly, nor provided whether they already have access to this model and tested it, I think it's a good idea to close this as "not planned" for the time being.

@HA-Ath0n if you have any additional information to provide, please reply it to this issue and we will reopen as necessary.

closed 2025-01-06 by MarijnS95