robbinjanssen/home-assistant-omnik-inverter #348: Omnik not showing "current power production" in energy dashboard

ekanters71, 2025-01-05:
## Detailed description
This is a bug I think. I have an Omnik 6K converter. It shows on the dashboard:
- Alarm Code - Unknown
- Current Power Production
- Solar Production Today
- Solar Production Total
- Solar Production Uptime - Unknown 

When configuring the energy dashboard I am not able to select the current power production, only the Solar Production Today or Total. I think I should have the Current Power Production also listed but it is not available.

## Context
In order to be able to calculate the energy use by the house I would need this data.

## Diagnostics
For some strange reason the status.html page only shows lines with +'es at the left. Correct values are showing up for today and totals.

```json
"Copy here the code or upload the file"
```

robbinjanssen, 2025-01-06:
Did you config it using the HTML source? If so, please provide me with the complete status.html page if possible. We have not yet supported an omnik 6k inverter yet.

MarijnS95, 2025-01-06:
At the same time we'd love to hear if the TCP connection is working :+1:

ekanters71, 2025-01-06:
Html is working although the status page from the omnik is not readable.

I think I misunderstood the energy picture. It is not actual but cumulatieve for the day. That nummer seems to be correct and increasing during the day.

My bad.

Regards,
Erwin Kanters

MarijnS95, 2025-01-10:
Sounds like this was resolves without actionable points on the repository side, so closing this as not-planned :+1:

closed 2025-01-10 by MarijnS95