robbinjanssen/home-assistant-omnik-inverter #349: Failed to connect Omnik 2500 TL2

mennes2000, 2025-01-09:
I installed the integration via HACS, but it won't work. If I follow the instructions, it says Failed to connect:
[image]

After logging in to my inverter:
[image]

And the JSON file via http://192.168.2.54/js/status.js:
`var version="H4.01.38Y1.0.09W1.0.07";var m2mMid="1607792926";var wlanMac="AC:CF:23:19:EF:C8";var m2mRssi="55%";var wanIp="192.168.2.54";var nmac="ACCF2319EFCB";var fephy="off";var webData="XXX,,V0.0-0,omnik2500tl2,2500,840,290,29,,0,";function initPageText(){var A=window.parent.reList("status");fileText("st1",A["t1"]);fileText("st2",A["t2"]);fileText("st3",A["t3"]);for(var B=1;B<=27;B++){if(B!=14){fileText("tx"+B,A[B])}}init_main_page();initStatus();changeFont();child_getH()}function upfold(A){if(document.getElementById("up_"+A+"_div").style.display=="none"){show("up_"+A+"_div");document.getElementById("up_"+A).src="image/up.png"}else{hide("up_"+A+"_div");document.getElementById("up_"+A).src="image/down.png"}if(A==3){if(document.getElementById("ser_status").src==""){document.getElementById("ser_status").src="server_status.html"}}}function init_main_page(){var E=document.form_cover_setting;cfg2Form(E);var C;var D;var B;var A;B=window.parent.reTip("1");A=window.parent.reTip("2");document.getElementById("cover_ver").innerHTML=version;document.getElementById("cover_mid").innerHTML=m2mMid;if(E.wifi_on.value=="disable"){document.getElementById("cover_ap_status").innerHTML=A;document.getElementById("cover_ap_mac").innerHTML="";document.getElementById("cover_ap_ssid").innerHTML="";document.getElementById("cover_ap_ip").innerHTML="";document.getElementById("cover_sta_status").innerHTML=A;document.getElementById("cover_sta_rssi").innerHTML="";document.getElementById("cover_sta_ip").innerHTML="";document.getElementById("cover_sta_mac").innerHTML="";document.getElementById("cover_sta_ssid").innerHTML=""}else{if(E.sta_enable.value=="1"){document.getElementById("cover_ap_status").innerHTML=B;document.getElementById("cover_ap_mac").innerHTML=wlanMac;document.getElementById("cover_ap_ssid").innerHTML=E.ap_ssid.value;document.getElementById("cover_ap_ip").innerHTML=E.lan_ip.value;document.getElementById("cover_sta_status").innerHTML=B;document.getElementById("cover_sta_rssi").innerHTML=m2mRssi;document.getElementById("cover_sta_ip").innerHTML=wanIp;C=wlanMac.substring(0,16);D="0x"+wlanMac.charAt(16);D=D*1+1;D=D.toString(16);D=D.toUpperCase();C=C+D;document.getElementById("cover_sta_mac").innerHTML=C;document.getElementById("cover_sta_ssid").innerHTML=E.sta_ssid.value}else{document.getElementById("cover_ap_status").innerHTML=B;document.getElementById("cover_ap_mac").innerHTML=wlanMac;document.getElementById("cover_ap_ssid").innerHTML=E.ap_ssid.value;document.getElementById("cover_ap_ip").innerHTML=E.lan_ip.value;document.getElementById("cover_sta_status").innerHTML=A;document.getElementById("cover_sta_rssi").innerHTML="";document.getElementById("cover_sta_ip").innerHTML="";document.getElementById("cover_sta_mac").innerHTML="";document.getElementById("cover_sta_ssid").innerHTML=""}}}function init_webdata_form(){var E=webData;var B="";var A,C,D;A=E.length;C=0;D=0;while(C<A){if(E.charAt(C)==","){C+=1;D+=1;web_data_show(D,B);B=""}else{B=B+E.charAt(C);C+=1}}}function web_data_show(D,A){var C=new Array("webdata_sn","webdata_msvn","webdata_ssvn","webdata_pv_type","webdata_rate_p","webdata_now_p","webdata_today_e","webdata_total_e","webdata_alarm","webdata_utime");if(D>10||D<=0){return}var B=document.getElementById(C[D-1]);if(A==""&&B!=null){B.innerHTML="---";return}if(B!=null){if(D==5){B.innerHTML=A+" W"}else{if(D==6){B.innerHTML=A+" W"}else{if(D==7){sv=A*1/100;B.innerHTML=sv.toString()+" kWh"}else{if(D==8){sv=A*1/10;B.innerHTML=sv.toString()+" kWh"}else{if(D==10){if(document.getElementById("webdata_sn").innerHTML=="---"){B.innerHTML="---"}else{B.innerHTML=A+window.parent.reTip("5")}}else{B.innerHTML=A}}}}}}else{alert(C[D])}};`

Am I doing something wrong, or is this inverter not supported (yet)?

mennes2000, 2025-02-10:
Currently running v2.6.4 (latest version), but it doesn't work.

camstar01, 2025-02-15:
Same here. I have two Omnik, worked perfectly for  a long time, but after a repair of one of them, I can’t add it. Everything looks OK, but keeps saying failed to connect.

camstar01, 2025-02-15:
> Same here. I have two Omnik, worked perfectly for a long time, but after a repair of one of them, I can’t add it. Everything looks OK, but keeps saying failed to connect. The other one is working perfectly.

I found a problem I think. In the status page the inverter segment is empty.

camstar01, 2025-02-25:
The status page of the inverter. There was no info on the inverter in the web page of the inverter. Power off and on solved it for a few days, but it is empty again So no integration problem, but a inverter problem

closed 2025-04-05 by github-actions[bot]